spuder/OpenSpool #2: AMS?

officiallymarky, 2024-11-25:
Does this have the ability to work with an AMS and different slots?
I'm wondering if this could be integrated to a system like the Python AMS.

spuder, 2024-11-25:
Prototype is currently in the works for AMS.

ser-attachments/assets/c3e97f24-3913-42aa-bcdb-ac4aa30701e8)

andysmithfal, 2024-11-25:
Interesting! I was imagining a separate unit to scan the material and select which AMS slot you're loading it into, but a fully integrated solution would be really tidy.

spuder, 2024-11-25:
The openspool mini is done but I have a few software bugs to finalize before I upload it to makerworld

koffienl, 2024-12-04:
Looks really promising. Just ordered the hardware parts. Can't wait for the AMS adapter :)

raihei, 2025-01-03:
Looks great. 

A second/alternate solution came to my mind, which needs less hardware and installation in AMS: 
Just 5 (touch) buttons on the mini to select, in which AMS slot (5th button for manual feeding) the spool will be put in. You can maybe even use only one button and press it multiple times to select the desired slot.

Workflow:
- Select desired slot by touching the corresponding button on the mini. -> LED confirmation (n flashes for the selected slot)
- Scan the spool -> LED will turn green and then pulse in a special color (e.g. yellow) until spool is installed and button is pressed again (or timeout after ~60-120 seconds)
- Put the spool into the AMS (or the manual feed)
- Press the corresponding button again to confirm, after AMS has finished loading -> LED confirmation
- Spool data will be transfered to the printer.

I don't have many experience with esphome, would that be even possible?

mytechguyri, 2025-01-08:
Would be great to develop something that works with Python and Hydra AMS as well.

koffienl, 2025-01-09:
> Looks great.
>
> A second/alternate solution came to my mind, which needs less hardware and installation in AMS: Just 5 (touch) buttons on the mini to select, in which AMS slot (5th button for manual [...]

I hope that this would only be an extra option. If I have to do all these manual steps I could just use the screen of my pandatouch.
I was in the understanding that for a AMS 2 readers would be enough, but looking at the render I would need 4? 
The ESP should already be aware of which slot is being load based on the MQTT info.

This weekend my buddy and I will attempt to assemble the components for the AMS, excited to wait how it will turn out.

spuder, 2025-01-09:
> I was in the understanding that for a AMS 2 readers would be enough

Bambu only uses 2 readers. The 2 readers are enough if the spools are placed perfectly, as it can figure out which slot is being loaded. 

The problem arrises when two adjacent spools both have an RFID tag next to the reader. Bambu does something clever and moves the motors until one of the RFID tags disappears. You may notice this occasionally when loading your AMS. 

Since we don't have the ability to control the motors directly, using 4 readers is a compromise. 
Also while the PN532 is physically capable of reading 2 tags at once, the esphome library doesn't support that yet.

raihei, 2025-01-09:
> Bambu only uses 2 readers. The 2 readers are enough if the spools are placed perfectly, as it can figure out which slot is being loaded.
>
> The problem arrises when two adjacent spools both have an [...]

Ah, that makes perfect sense. I was already wondering, why you planned to use 4 readers.

koffienl, 2025-01-11:
> Bambu only uses 2 readers. The 2 readers are enough if the spools are placed perfectly, as it can figure out which slot is being loaded.
>
> The problem arrises when two adjacent spools both have an [...]

Where can I find wiring schematics to build the AMS version with 4 readers?