spuder/OpenSpool #6: Both 5V and 3V are hooked up?

RobThree, 2024-12-07:
I'm confused. Why is **both** 3V and 5V hooked up?

[image]

Isn't it enough to hook up just 5V? It would save a levelshifter (or even a 4-levelshifter module that could be a single levelshifter module).

spuder, 2024-12-07:
The level shifter is only for the data line of the led. Since the LED needs 5v logic but the esp32 GPIOs only provide 3.3v some LEDS are unreliable with the lower voltage. 

On the PCB version, the 4 line shifter has been replaced with a single transistor and 2 resistors. 
[image]

RobThree, 2024-12-07:
> The level shifter is only for the data line of the led. Since the LED needs 5v logic but the esp32 GPIOs only provide 3.3v some LEDS are unreliable with the lower voltage.

I wasn't asking what the levelshifter is for. In the diagram you're hooking up **both** 5V (bottom right) and 3V (bottom left) to the Wemos D1, See the arrows. That doesn't make sense. And _that_ makes the "second" levelshifter redundant.

RobThree, 2024-12-07:
Wait, wait, wait... I'm a dumbass....

You're correct. You need to get the 3V from somewhere. It's coming FROM the Wemos TO (feed) the levelshifter.... 

Jeez. My bad.

closed 2024-12-07 by RobThree

spuder, 2024-12-07:
The quad level shifter is overkill. 
The single level shifters I found on amazon were actually more expensive than the quads so I linked the cheaper versions. 

Here is the schematic I'm using. It seems to work, but if it can be improved I'll gladly change it. 

[image]