spuder/OpenSpool #12: FR: Spool SN field

GunbleR, 2024-12-16:
Havent tried yet, may be usefull to add optional serial number to tag to differanciate between old and new simmilar spools.
Mainly for spool managers if they decide to implement support for it

spuder, 2024-12-16:
Thanks for the idea. 

As I don't have much experience with spool-man or any other databases, what might this look like? 

Is a serial number unique per spool or unique per rfid tag? 

The tags already have a UID that could be used.

AA-BB-CC-DD (mifare classic)
AA-BB-CC-DD-EE-FF (ultralight/NTAG)

bofh69, 2024-12-18:
In Spoolman you create a filament and then from the filament you can create one or more spools. Each filament and each spool get their own ID (just an integer right now). When you use Spoolman with OctoPrint or Klipper & Moonraker you need to tell them which spool is in use by using Spoolman's spool ID. They then use it to report the filament usage to Spoolman.

Spoolman has a REST API that can do almost everything the web interface can do. You can configure Spoolman with custom fields for both its filament and spool records as needed.

With nfc2klipper, I first create new spools in Spoolman, then create the tags from Spoolman's data. The tags only contain the filament and spool IDs. The filament ID is used in my gcode to verify the right filament is used and ask me to change spool if the wrong filament is loaded (thus I can change to new spools with the same material and the verification still passes).

If you create the tags first, you can lookup the filament id in Spoolman (or create it if missing with the data from the tag. You would need to find the vendor/manufacturer too for that), then create a spool and there use a custom field for the UID.

You can't search in Spoolman via custom fields, but you can get all spools matching a certain filament name, material, vendor/manufacturer, or simply get all spools and then search for it in that way.

thijstriemstra, 2025-02-04:
Duplicate of #9