spuder/OpenSpool #13: 3DFilamentProfiles.com as source

domwrap, 2024-12-17:
I thought I found somewhere the data feed you were using for the source input to choose brand/filament but now can't find it.

Would love to see integration from https://3dfilamentprofiles.com/ which has profiles for most filaments (and is where a lot of use to print QR codes for filament storage and organization) so could be used to populate the options on the tag config screen. Even if Bambu doesn't currently recognize or have profiles for many of them, this could at least be used as the source data to write the tags, and when reading the tag and updating the Bambu printer re-map it at that stage if necessary.

Absolutely love this project, following along closely and will probably be trying to build a kit in the new year. Kudos. 👍

spuder, 2024-12-18:
Thanks for sharing https://3dfilamentprofiles.com/. This is a great resource. People have asked me if there is a centralized location for profiles and this likely is the best location. 

As far as pulling this resource automatically into openspool it might be a bit tricky as the esp32 needs a significant amount of PSRAM to establish https connections. An esp32-s3 should be able to do it, but an esp32-s2 likely won't. 

There is also the issue that bambu uses an internal mapping of hex values and color names. If providing a value that isn't in their predefined list, it will show a blank color on the screen. 

OpenSpool plans to expand out to printers outside of bambu, so for non bambu printers this likely will be very useful..

markcarroll, 2024-12-20:
Hi @spuder I am the developer of the 3D Filament Profiles site.  Let me know if you want to discuss best path to integration.

spuder, 2024-12-21:
👋 Hi @markcarroll Thanks for reaching out. 

Yes I would be interested in discussing more how we could use your site as a single source of truth for filaments. 
The esp32 has only 320k of ram so trying to parse the entire database might not be feasible. 

Is there a way to export the "My Spools" feature to a json file or pull from a rest api? 

I envision a text box on the openspool web interface where a user could enter an apikey or credential. It then downloads that users "My Spools" selection and adds them to a dropdown box they could choose from.

huizebruin, 2024-12-21:
Hello @spuder and @markcarroll .

Why not to make a combination of three.
i am no programmer, but i think this would bee the best to do, because the memory inside a esp32 isnt very big.

The openspool connected to spool manager and thisone gets also the data form 3dfilamentprofiles?
becase the spoolman has also a api https://donkie.github.io/Spoolman/#operation/
i dont no if this is possible but than you have a big and strong setup i think.

https://github.com/Donkie/Spoolman and https://github.com/Disane87/spoolman-homeassistant
thatone works also inside home assistant as a addon,and works with [OctoPrint](https://octoprint.org/) and [Klipper](https://www.klipper3d.org/)/[Moonraker](https://moonraker.readthedocs.io/en/latest/). for connecting to your printer.

markcarroll, 2024-12-21:
@huizebruin I am working on HA and printer integration to 3DFP's My Spools too, but don't want to go off topic here. 

@spuder what if I put a link on the 3DFP filament details page that downloads the data you need to create a tag?  Or I can give you an API that gives you that same data. Given the size of the database now, doing the choosing on the web site end would probably be more efficient.  Or maybe we can go to the filament selector page and then do a callback with the JSON.

Without one of your devices I am not super clear on how the mechanics of your side works.  Given some more insight into your system I can happily help work out a good integration. You can get me on Reddit: u/SirThunderCloud or Discord: @sir.thunder.cloud if you want to chat.

koffienl, 2025-01-22:
I was thinking, how about an API link from the 3dfilaments filament details website to the openspool GUI on the ESP?
You open a specific filament on the 3dfilaments site and where the QR code is there is also a link to callback to the OpenSpool GUI to POST or GET the info in JSON.
The GUI opens, with all the necessary info and all you have to do is click write tag.
Alternative you could do it through the openspool.io site, but a quick search tell me writing NFC on mobile through a website has some security issues.

Roving-Ronin, 2025-01-22:
Why not use the OpenSpool application, it has an API you can call that sends back info from its OpenspoolDB (filaments.json).
Easy as to add new filaments via its GitHub.

DocJ-GA, 2025-05-17:
I am working on a spoolman to orcaslicer console app that syncs spoolman spools to orcaslicer filaments.  I would love to look at developing an addon to spoolman to sync extra settings from 3dfilamentprofiles to spoolman via an api.