spuder/OpenSpool #14: MQTT limit

demonkazuya, 2024-12-27:
Is there a way we can have this integrated with home assistant? I read that the machine can handle a certain amount of MQTT. I think the reason why I get disconnected is because I have several devices connected (Home Assistant, Xtouch). Not sure if that's the reason for disconnection. I know this is too much of an ask, but it would be great it can be added. Thanks!

spuder, 2024-12-27:
I see you are having disconnects. When you do get a disconnect are there any messages in the logs? 

What processor are you using? esp32-s2 or esp32-s3? 

What printer type? (A1 Mini, A1, P1P, P1S, X1)

If running the s2, note that it has less memory than the s3 so disconnects are a side effect. Consider upgrading to an s3 board. 

I've not yet confirmed but supposedly certain printers allow less MQTT subscribers than others. (Allegedly the limit is 4, but higher on the X1 models). 

As far as HomeAssistant, it should be possible to setup as a MQTT Bridge where HA talks to the printer and all your devices talk to HA.

demonkazuya, 2024-12-27:
it's random actually. I'm using the S2.. will try to use S3 and test. It's random actually was not able to have the logs. My printer is a P1S

Also thank you for the suggestion will look at the MQTT bridge to see if that would help. Thanks!

Will keep you posted with the testing

demonkazuya, 2024-12-27:
Here are the logs

`01:10:06
[W]
[mqtt:340]
MQTT Disconnected: TCP disconnected.
01:10:06
[I]
[mqtt:250]
Connecting to MQTT...
01:10:21
[V]
[mqtt.idf:121]
Event dispatched from event loop event_id=7
01:10:21
[V]
[mqtt.idf:124]
MQTT_EVENT_BEFORE_CONNECT
01:10:22
[V]
[mqtt.idf:121]
Event dispatched from event loop event_id=0
01:10:22
[E]
[mqtt.idf:162]
MQTT_EVENT_ERROR
01:10:22
[E]
[mqtt.idf:164]
Last error code reported from esp-tls: 0x8017
01:10:22
[E]
[mqtt.idf:165]
Last tls stack error number: 0x7f00
01:10:22
[E]
[mqtt.idf:166]
Last captured errno : 0 (Success)
01:10:22
[V]
[mqtt.idf:121]
Event dispatched from event loop event_id=2
01:10:22
[V]
[mqtt.idf:133]
MQTT_EVENT_DISCONNECTED
01:10:22
[I]
[main:764]
Disconnected from printer!
01:10:22
[W]
[mqtt:340]
MQTT Disconnected: TCP disconnected.
01:10:22
[I]
[mqtt:250]
Connecting to MQTT...
01:10:36
[V]
[mqtt.idf:121]
Event dispatched from event loop event_id=7
01:10:36
[V]
[mqtt.idf:124]
MQTT_EVENT_BEFORE_CONNECT
01:10:36
[V]
[mqtt.idf:121]
Event dispatched from event loop event_id=0
01:10:36
[E]
[mqtt.idf:162]
MQTT_EVENT_ERROR
01:10:36
[E]
[mqtt.idf:164]
Last error code reported from esp-tls: 0x8017
01:10:36
[E]
[mqtt.idf:165]
Last tls stack error number: 0x7f00
01:10:36
[E]
[mqtt.idf:166]
Last captured errno : 0 (Success)
01:10:36
[V]
[mqtt.idf:121]
Event dispatched from event loop event_id=2
01:10:36
[V]
[mqtt.idf:133]
MQTT_EVENT_DISCONNECTED
01:10:36
[I]
[main:764]
Disconnected from printer!
01:10:36
[W]
[mqtt:340]
MQTT Disconnected: TCP disconnected.
01:10:36
[I]
[mqtt:250]
Connecting to MQTT...
01:10:51
[V]
[mqtt.idf:121]
Event dispatched from event loop event_id=7
01:10:51
[V]
[mqtt.idf:124]
MQTT_EVENT_BEFORE_CONNECT
01:10:52
[V]
[mqtt.idf:121]
Event dispatched from event loop event_id=0
01:10:52
[E]
[mqtt.idf:162]
MQTT_EVENT_ERROR
01:10:52
[E]
[mqtt.idf:164]
Last error code reported from esp-tls: 0x8017
01:10:52
[E]
[mqtt.idf:165]
Last tls stack error number: 0x7f00
01:10:52
[E]
[mqtt.idf:166]
Last captured errno : 0 (Success)
01:10:52
[V]
[mqtt.idf:121]
Event dispatched from event loop event_id=2
01:10:52
[V]
[mqtt.idf:133]
MQTT_EVENT_DISCONNECTED
01:10:52
[I]
[main:764]
Disconnected from printer!
01:10:52
[W]
[mqtt:340]
MQTT Disconnected: TCP disconnected.
01:10:52
[I]
[mqtt:250]
Connecting to MQTT...
01:11:06
[V]
[mqtt.idf:121]
Event dispatched from event loop event_id=7
01:11:06
[V]
[mqtt.idf:124]
MQTT_EVENT_BEFORE_CONNECT
01:11:06
[V]
[mqtt.idf:121]
Event dispatched from event loop event_id=0
01:11:06
[E]
[mqtt.idf:162]
MQTT_EVENT_ERROR
01:11:06
[E]
[mqtt.idf:164]
Last error code reported from esp-tls: 0x8017
01:11:06
[E]
[mqtt.idf:165]
Last tls stack error number: 0x7f00
01:11:06
[E]
[mqtt.idf:166]
Last captured errno : 0 (Success)
01:11:06
[V]
[mqtt.idf:121]
Event dispatched from event loop event_id=2
01:11:06
[V]
[mqtt.idf:133]
MQTT_EVENT_DISCONNECTED
01:11:06
[I]
[main:764]
Disconnected from printer!
01:11:06
[W]
[mqtt:340]
MQTT Disconnected: TCP disconnected.
01:11:06
[I]
[mqtt:250]
Connecting to MQTT...`

so I notice it connects on start up but does this after sometime

Maxklos, 2025-03-10:
Hey, I get the same error and OpenSpool won't connect to my printer:

> A1Mini
> ESP32-S3

```18:19:06[I][mqtt:250]Connecting to MQTT...
18:19:06[V][mqtt.idf:121]Event dispatched from event loop event_id=7
18:19:06[V][mqtt.idf:124]MQTT_EVENT_BEFORE_CONNECT
18:19:06[V][mqtt.idf:121]Event dispatched from event loop event_id=0
18:19:06[E][mqtt.idf:162]MQTT_EVENT_ERROR
18:19:06[E][mqtt.idf:164]Last error code reported from esp-tls: 0x8017
18:19:06[E][mqtt.idf:165]Last tls stack error number: 0x7f00
18:19:06[E][mqtt.idf:166]Last captured errno : 0 (Success)
18:19:06[V][mqtt.idf:121]Event dispatched from event loop event_id=2
18:19:06[V][mqtt.idf:133]MQTT_EVENT_DISCONNECTED
18:19:06[I][main:781]Disconnected from printer!
18:19:06[W][mqtt:340]MQTT Disconnected: TCP disconnected.
```

The A1Mini is connected to the Bambu App & Home Assistant nothing else.

serkan887, 2025-03-14:
Same here. p1s connected just homeassistant, and esp32-s2 board.

`21:56:47
[I]
[mqtt:250]
Connecting to MQTT...
21:57:02
[V]
[mqtt.idf:121]
Event dispatched from event loop event_id=7
21:57:02
[V]
[mqtt.idf:124]
MQTT_EVENT_BEFORE_CONNECT
21:57:02
[V]
[mqtt.idf:121]
Event dispatched from event loop event_id=0
21:57:02
[E]
[mqtt.idf:162]
MQTT_EVENT_ERROR
21:57:02
[E]
[mqtt.idf:164]
Last error code reported from esp-tls: 0x8017
21:57:02
[E]
[mqtt.idf:165]
Last tls stack error number: 0x7f00
21:57:02
[E]
[mqtt.idf:166]
Last captured errno : 0 (Success)
21:57:02
[V]
[mqtt.idf:121]
Event dispatched from event loop event_id=2
21:57:02
[V]
[mqtt.idf:133]
MQTT_EVENT_DISCONNECTED
21:57:02
[I]
[main:781]
Disconnected from printer!
21:57:02
[W]
[mqtt:340]
MQTT Disconnected: TCP disconnected.
21:57:02
[I]
[mqtt:250]
Connecting to MQTT...`