spuder/OpenSpool #15: Would it be possible to use the I2C interface instead of the SPI interface?

pzw001, 2024-12-29:
Hello,

I am very interested in your project, and am currently looking at the creation of a PCB which will include everything needed in a nice organized manner. My goal would be at this moment to use the project for my 2 AMS units, so I would need 8 (4 if I can get them to read at both sides ;-) ) or 9 (5) readers. I see that the current solution is to use designated IO pins for the CS signals, which is the easiest / quickest way. But there might be easier options using the I2C interface instead. (Especially with 8 or more readers connected). The goal I set for myself is to get the PCB to fit in one of the openings in the AMS enclosure, so it looks like a nice finished unit.

I would be happy to share my PCB design with the community, once all the possible issues have been resolved. 

It might be best to discuss this in an email message instead. Please let me know.

Thanks
Paul

spuder, 2024-12-29:
Hi thanks for reaching out.

So I understand you have 2 AMS and you want to design a PCB that supports 8 readers.

Your help would be greatly appreciated. Here is the current status:

PCB

I have a hand soldered prototype that supports 5 readers. It uses a more powerful esp32-s3. I've also sourced the exact molex connectors that the AMS uses so it could be powered from the 24v from the printer instead of USB.
Designs are in EasyEda. I can invite you to the project. 

3d prints 

There is an experimental 3d printed insert that fits inside an AMS. It uses DuPont connectors but likely it should be changed to JST connectors. Appropriate cables would need to be sourced

SPI/i2c

I2c is a good idea, however it's not feasible for the project for a couple of reasons.
1. Every pn532 that is sourced has a hard coded 0x42 address with no way to change it without a hardware multiplexer.
2. Clock stretching
I tried for many days to get i2c working with no luck. I found lots of several of people having the same problems. Those who did get it working found it unreliable. The pn532 uses clock stretching which the Esphome library doesn't support. 

Since the p1/x1 support up to 17 spools, the preferred design will probably be to have 1x esp32 _per AMS_ and use a CANBUS or wireless link between them, where one of them. Since the P1 (and possibly the x1) have a limit of 4 mqtt clients. More prototypes are needed. 

You can reach me in the discord channel as user @spuder and we can chat more if you like.

spuder, 2024-12-29:
[image]

spuder, 2024-12-29:
[image]

koffienl, 2025-01-01:
> ![image](https://private-user-images.githubusercontent.com/242382/399180060-76f0ffc1-7729-49f7-b4bf-a2c6b0e53558.jpeg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzU3MzQ1OTYsIm5iZiI6MTczNTczNDI5NiwicGF0aCI6Ii8yNDIzODIvMzk5MTgwMDYwLTc2ZjBmZmMxLTc3MjktNDlmNy1iNGJmLWEyYzZiMGU1MzU1OC5qcGVnP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI1MDEwMSUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNTAxMDFUMTIyNDU2WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9ZDRjNTVjNDEzYTMxMzQ0Nzg2YTFjMGIxYjAyZDc0Yzc1OTYzNmU3MGZmYTk4NmYxZGNmODMzODAwZmQzZjdmNSZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QifQ.sZW3Dju977V1GV7mHe4eCcXoLP9PxY8wDgbu7AQBmOg)

Any chance this would be available for download any time soon?

ManuelW77, 2025-02-01:
> Since the p1/x1 support up to 17 spools, the preferred design will probably be to have 1x esp32 _per AMS_ and use a CANBUS or wireless link between them, where one of them is the dedicated mqtt client since the P1 (and possibly the x1) have a limit of 4 mqtt clients.

You can use a simple mqtt proxy for that. the proxy connects to the printer and copys the channels read/write. So you can connect as many clients you want to the proxy.