spuder/OpenSpool #16: Component pn532 is marked FAILED

CyberFreek, 2024-12-30:
Hi, I am trying to setup an OpenSpool Mini and purchased the components and connected everything as per the diagram on the site. My issue is that it seems to be giving me an error related to the PN532 RFID reader, but it doesn't really say what the error is for. Is there any  way I can get a more detailed error?

There is a solid red led illumined on the PN532, but I have no indication what it means.

I am using a Wemos D1 Mini s3.

Here are my logs - 

```
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x15 (USB_UART_CHIP_RESET),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x4037bd4e
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce2810,len:0x178c
load:0x403c8700,len:0x4
load:0x403c8704,len:0xc10
load:0x403cb700,len:0x2dc0
entry 0x403c88fc
I (26) boot: ESP-IDF 5.3.1 2nd stage bootloader
I (27) boot: compile time Dec 14 2024 20:33:11
I (27) boot: Multicore bootloader
I (27) boot: chip revision: v0.1
I (27) boot.esp32s3: Boot SPI Speed : 80MHz
I (27) boot.esp32s3: SPI Mode       : DIO
I (28) boot.esp32s3: SPI Flash Size : 4MB
I (28) boot: Enabling RNG early entropy source...
I (28) boot: Partition Table:
I (28) boot: ## Label            Usage          Type ST Offset   Length
I (29) boot:  0 otadata          OTA data         01 00 00009000 00002000
I (29) boot:  1 phy_init         RF data          01 01 0000b000 00001000
I (29) boot:  2 app0             OTA app          00 10 00010000 001c0000
I (30) boot:  3 app1             OTA app          00 11 001d0000 001c0000
I (30) boot:  4 nvs              WiFi data        01 02 00390000 0006d000
I (31) boot: End of partition table
I (31) esp_image: segment 0: paddr=00010020 vaddr=3c0e0020 size=46724h (288548) map
I (83) esp_image: segment 1: paddr=0005674c vaddr=3fc9a600 size=04918h ( 18712) load
I (87) esp_image: segment 2: paddr=0005b06c vaddr=40374000 size=04fach ( 20396) load
I (92) esp_image: segment 3: paddr=00060020 vaddr=42000020 size=d6b28h (879400) map
I (250) esp_image: segment 4: paddr=00136b50 vaddr=40378fac size=11640h ( 71232) load
I (276) boot: Loaded app from partition at offset 0x10000
I (276) boot: Disabling RNG early entropy source...
I (277) cpu_start: Multicore app
I (286) cpu_start: Pro cpu start user code
I (286) cpu_start: cpu freq: 160000000 Hz
I (286) app_init: Application information:
I (286) app_init: Project name:     openspool
I (286) app_init: App version:      2024.11.3
I (287) app_init: Compile time:     Dec 14 2024 20:32:23
I (287) app_init: ELF file SHA256:  ad730b13f...
I (287) app_init: ESP-IDF:          5.3.1
I (287) efuse_init: Min chip rev:     v0.0
I (287) efuse_init: Max chip rev:     v0.99 
I (287) efuse_init: Chip rev:         v0.1
I (288) heap_init: Initializing. RAM available for dynamic allocation:
I (288) heap_init: At 3FCA39B8 len 00045D58 (279 KiB): RAM
I (288) heap_init: At 3FCE9710 len 00005724 (21 KiB): RAM
I (289) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM
I (289) heap_init: At 600FE100 len 00001EE8 (7 KiB): RTCRAM
I (290) spi_flash: detected chip: generic
I (290) spi_flash: flash io: dio
W (291) rmt(legacy): legacy driver is deprecated, please migrate to `driver/rmt_tx.h` and/or `driver/rmt_rx.h`
I (293) sleep: Configure to isolate all GPIO pins in sleep state
I (293) sleep: Enable automatic switching of GPIO sleep configuration
I (294) main_task: Started on CPU0
I (295) main_task: Calling app_main()
I (407) main_task: Returned from app_main()
[I][logger:171]: Log initialized
[C][safe_mode:079]: There have been 2 suspected unsuccessful boot attempts
[... 185 lines cut ...]
[V][mdns:122]:     TXT: mac = f0f5bd6d4294
[V][mdns:122]:     TXT: platform = ESP32
[V][mdns:122]:     TXT: board = lolin_s3_mini
[V][mdns:122]:     TXT: network = wifi
[V][mdns:122]:     TXT: project_name = spuder.openspool
[V][mdns:122]:     TXT: project_version = 1.14.0
[V][mdns:122]:     TXT: package_import_url = github://spuder/openspool/firmware/lolin_s3_mini.yaml@main
[V][mdns:120]:   - _http, _tcp, 80
[C][esphome.ota:073]: Over-The-Air updates:
[C][esphome.ota:074]:   Address: openspool.local:3232
[C][esphome.ota:075]:   Version: 2
[C][safe_mode:018]: Safe Mode:
[C][safe_mode:019]:   Boot considered successful after 60 seconds
[C][safe_mode:021]:   Invoke after 10 boot attempts
[C][safe_mode:022]:   Remain in safe mode for 300 seconds
[W][safe_mode:028]: Last reset occurred too quickly; safe mode will be invoked in 8 restarts
[C][mqtt:148]: MQTT:
[C][mqtt:149]:   Server Address: 192.168.1.177:8883 (0.0.0.0)
[C][mqtt:151]:   Username: [redacted]
[C][mqtt:152]:   Client ID: [redacted]
[C][mqtt:153]:   Clean Session: YES
[C][mqtt:161]:   Topic Prefix: ''
``` 
It also won't let me enable the MQTT:
[image]

spuder, 2024-12-30:
Double check the PN532 has the pins configured for SPI mode.

bvestal-nlg, 2024-12-31:
Unless I am reading this wrong, I believe it is...

[image]

spuder, 2024-12-31:
That looks correct. What does the esp32 side look like?

CyberFreek, 2025-01-01:
So I picked up another pn532 on amazon yesterday (the other one was from Aliexpress) and wired it up - only to get the exact same error with all new wires. 

Here's the S3
[image]
[image]

I also use a fluke multimeter to verify that every wire is making a 100% connection and they are all good.

Has anyone else got the S3 working?

CyberFreek, 2025-01-01:
Oh and I also tried wiping the S3 and reinstalling, to no avail.

Is there some way to increase the detail in the logging for the pn532 to perhaps get some better errors?

spuder, 2025-01-01:
That's frustrating, sounds like you are doing everything right. Here are a couple of other ideas

1. Are you powering from 5v or 3.3v ? Both should work, however the PCBs use 5v if I recall
2. Is the PN532 getting hot? 
3. Have you ever hooked it up backwards? I've killed one by accidentally reversing the polarity. It never worked again and started experiencing exactly what you described
4. Do you have the ability to inspect the crystal on the PN532? I received a batch of PN532 that were counterfeit that used a 27.00 MHz crystal instead of the proper 27.12 MHz crystal. Seems unlikely to get 2 counterfeits from 2 vendors, but it is possible. 
5. Did you compile yourself or use the web installer at openspool.io? If compiling yourself you need to be careful to select the correct yaml file. The web installer should detect it correctly

If it can't talk to the PN532 then this is 90% most likely a hardware issue. 
If you do all this and still can't get it working let me know and I can send you a PCB.

CyberFreek, 2025-01-01:
Hi @spuder, thanks again for the reply! I am sure you have better things to do than to help me troubleshoot my issue, but I am extremely appreciative of your assistance.

 >     1. Are you powering from 5v or 3.3v ?

 I am powering everything from the USB-C port on the S3, which is 5v.

>     2. Is the PN532 getting hot?

Not noticeably, although I haven't left it plugged in for more than a couple minutes at a time.  Is it supposed to get hot?

>     3. Have you ever hooked it up backwards? 

No, although since this is starting to point to a hardware issue I am starting to question everything LOL. For example, I had a little trouble soldering the header onto the S3 and so it did get pretty toasty. I suppose it is possible that I stressed something too much and it is causing the errors. Or maybe a solar flare cause a CME and fried it. 

>     4. Do you have the ability to inspect the crystal on the PN532? 

Alas, the Fluke Multimeter is the fanciest tool in my inventory. I assume I would need something pretty expensive to be able to inspect the crystal like an oscilloscope. I am almost certain the finance department (my wife) would probably frown if something like that appeared on my desk since it is one of those things that just _looks_ expensive.

>     5. Did you compile yourself or use the web installer at openspool.io? 

 I have been using openspool.io exclusively. I had considered trying to compile it myself to see if there is any difference, but I am starting to think it really is a hardware issue. 

>      If it can't talk to the PN532 then this is 90% most likely a hardware issue. If you do all this and still can't get it working let me know and I can send you a PCB.

That is what I am thinking as well. I have quadruple checked that I have it wired exactly like your diagram shows, with the exception of the WAGO connectors. I didn't have any of those so I simply used a wire nut to connect the 4 5V wires and the 5 ground wires. I was operating under the assumption that was essentially the purpose of the Wago connectors, was that a correct assumption?

I very much appreciate the offer of the PCB, but at this point the only thing I haven't replaced is the S3 mini, so I went ahead and ordered another one of those from Aliexpress along with a PN532 from a different vendor - just to be on the safe side.

I guess the only other question I have is if I decide to get impatient waiting for Aliexpress, could I use something like [this](https://www.amazon.com/dp/B09R4GSDJM?psc=1&ref_=cm_sw_r_cp_ud_ct_Z67FC12AF4AHVW2P3FF4) instead of an S3 mini? 

I noticed in your Github that you also have a driver for the [ESP32-S3-DevKitC-1](https://github.com/spuder/OpenSpool/blob/main/firmware/esp32-s3-devkitc-1.yaml), which is something I could get a lot faster than waiting for AliExpress.

Finally, have you made the STL for the case available anywhere yet? It looks pretty cool and I would like to print one. 😉

And if you put it on Makerworld I have some boot tokens I can send your way too. 

Thanks again for all the help!

spuder, 2025-01-01:
Another thought. Which diagram are you using? There is an old schematic floating around that had different PIN numbers.

For an s3 these are the pins they need to be used. 

https://github.com/spuder/OpenSpool/blob/main/firmware/lolin_s3_mini.yaml

You are welcome to switch to a dev kit however it has different PIN numbers. 

Theoretically other s2 and s3 boards would work but you will need to manually figure out the pin mappings.

As far as checking the crystal, you just need a magnifying glass or microscope to look at the numbers on it.

CyberFreek, 2025-01-02:
> Another thought. Which diagram are you using? There is an old schematic floating around that had different PIN numbers.
>
> For an s3 these are the pins they need to be used. [...]

I used this diagram: https://github.com/spuder/OpenSpool/blob/main/images/OpenSpoolMiniWiringDiagram2.png

I took a look at that YAML file and the pins look correct. 

Ha!  Well I guess i was overcomplicating what it takes to check the crystal! 

Here's the original PN532: 
[image]

And here is the second one:
[image]

Both of which indicate 27.120 mhz. It was a good thought though!

I did notice something else though, even though I enter my Bambu credentials, I noticed this in that same logfile above:
[D][main:114]: Missing Bambu Credentials, skipping mqtt connect

When I go back and check, they appear to be saved in there correctly, it doesn't seem to be trying them for some reason. Also, in this screenshot, it doesn't show MQTT as enabled - nor can I click on the circle to enable it. 
[image]

Do you have any idea what might be going on here? I wouldn't think this is related to the PN532, but I'm not 100% sure.

Thanks again for the assistance!

spuder, 2025-01-02:
Is the serial number populated? That is also required for MQTT connections. 

There might be an ordering dependency that mqtt won't succeed if mqtt is failing. I've not tested that. I would first focus on getting past the `Component pn532 is marked FAILED` error. 

It seems unlikely that you got 2 counterfeit PN532 in a row. I have 100 of them in stock, I'll sell you a verified working one at cost if you want as I would really like to see you have a working kit.

CyberFreek, 2025-01-03:
> Is the serial number populated? That is also required for MQTT connections.
> 
Actually, I did not have the serial number entered. I did add it and reboot and appears to be looping. plus, the MQTT Connection is still uncheckable as seen below...
[image]
[image]
[image]

> There might be an ordering dependency that mqtt won't succeed if mqtt is failing. I've not tested that. I would first focus on getting past the `Component pn532 is marked FAILED` error.
> 

Much of this is new ground for me and being in IT for as many decades as I have I have come to learn that sometimes completely unrelated things are connected. I figured if that error was a simple fix, it wouldn't hurt to resolve it - but if you think their might be some dependency on the PN532 loading correctly then we can put that one on the back burner for now.

> It seems unlikely that you got 2 counterfeit PN532 in a row. I have 100 of them in stock, I'll sell you a verified working one at cost if you want as I would really like to see you have a working kit.

Yea, my gut tells me the same thing - that it is highly unlikely that both PN532's are bad - especially since the second one I tried seems to be something of a brand - at least enough to have a website with downloadable manuals [here](https://www.elechouse.com/elechouse/images/product/PN532_module_V3/PN532_%20Manual_V3.pdf). 

Still, if your gut is telling you that this feels more like a PN532 issue than a S3-mini issue, then sure I don't mind giving one of yours a try. How should I contact you for payment, address, etc?

CyberFreek, 2025-01-04:
BTW, I just noticed you posted to model out on Makerworld and you are already getting some love, which is awesome!

spuder, 2025-01-04:
If you join the discord and message me I'll send you a board. 

https://discord.gg/4EaXHu9CEj

User name is `@spuder`

CyberFreek, 2025-01-09:
So I picked up one of [these](https://www.amazon.com/gp/product/B0CT5DJK7H/ref=ppx_yo_dt_b_asin_title_o00_s00?ie=UTF8&psc=1) from Amazon and realized that the default PIN layout wasn't going to work. So I pulled a copy of the OpenSpool repo, updated a couple of file (in the OpenSpool folder of the ZIP, the Makefile and the YAML that included new GPIO pins as well as support for the WinShare ESP32-S3-Zero card). Then i realized I needed to recompile the firmware.
[esp32-s3-zero.zip](https://github.com/user-attachments/files/18354433/esp32-s3-zero.zip)
So then I fired up a docker container with ESPHome on it and updated the PlatformIO.ini file in the root of the docker image as well as the new board JSON in the .PlatformIO folder located in the home folder and was able to successfully compile and flash the firmware!

It turns out the PN532 board were all fine. The issue was with the S3-Mini. I'm not sure if I just burned it with the soldering iron or what but as you can see here, it found my PN532 and I have been able to write and scan NFC tags with it. 
[image]

Honestly, I"m not sure if I did all that the right way or not but it got my board working! 

I don't suppose there is some way to integrate these updated files with what's on Openspool.io? That way when new firmware is released it will be a bit easier for me to update.

closed 2025-01-09 by CyberFreek

spuder, 2025-01-09:
Awesome! So glad to hear. Very strange that your board didn't work.

impeeza, 2025-11-08:
> Zero

Mate sorry for contact you, hope you do not mind.

I was only able to get a Waveshare-S3-Zero, I really do not know how to use docker (I know I am a dinosaur)  Wonder if you can share your build with me while I try to learn how to launch the docker.  did you launched local on your pc or used a cloud service?

Thanks a lot.