spuder/OpenSpool #18: Filament color temperatures are not inutitive

spuder, 2025-01-06:
As a user who wants to program a tag
It is confusing that fields which can't be selected (Filament Color Hex) are next to fields that must be selected ( temperature, type)

[image]

## Requested Changes

Re-group the "Filament Settings" section so that user specified fields are at the top, and informational fields are below

- Filament Brand
- Filament Color
- Filament Max Temp
- Filament Min Temp
- Filament Type

---

- Filament Color Hex
- Filament Color Hex Bambu (maybe just remove this line entirely)

The "Upload Settings Button" and "Write NFC" Buttons should probably be moved to be more intuitive.

raihei, 2025-01-12:
Do you want to create a new group for the the informational fields? I think, moving them down already makes it far more clearer without "wasting" more space for an additional group header. I prefer, to have all 4 drop down menus at the top, and then the sliders for the temperature. It looks more clean, and makes no difference in sense.
[image]

[image]

spuder, 2025-01-12:
> Do you want to create a new group for the the informational fields?

I like your idea. A new group is probably overkill. Simply re-sorting the fields like you have shown would be cleaner if that is possible. 

I haven't sorted entities within a group but it looks like the web server component allows sorting by both group and entity. 

https://esphome.io/components/web_server.html#entity-sorting

raihei, 2025-01-12:
Thanks. 
Yes, that is how I have done it, I have just added a sorting_weight to each entity.
I can create a pull request, if wanted.

spuder, 2025-01-12:
> I can create a pull request, if wanted.

Yes please. 

We are going to record a tutorial video this week of how to use the interface, so your timing is perfect. I this gets merged in before the next couple of days then we will include your changes in the video.

spuder, 2025-01-12:
Released in v1.15.0

closed 2025-01-12 by spuder