spuder/OpenSpool #20: Enhancement Request:  Pass alpha channel in RGB code to allow for transparent filaments

mytechguyri, 2025-01-09:
I see in the Bambu MQTT API, that they actually support the alpha channel in the MQTT message as shown here

`{
    "print": {
        "sequence_id": "0",
        "command": "ams_filament_setting",
        "ams_id": 0, // Index of the AMS
        "tray_id": 0, // Index of the tray
        "tray_info_idx": "", // Probably the setting ID of the filament profile
        _**"tray_color": "00112233", // Formatted as hex RRGGBBAA (alpha is always FF)**_
        "nozzle_temp_min": 0, // Minimum nozzle temp for filament (in C)
        "nozzle_temp_max": 0, // Maximum nozzle temp for filament (in C)
        "tray_type": "PLA" // Type of filament, such as "PLA" or "ABS"
    }
}`

Sending the alpha channel as 00 instead of FF tells Bambu Studio its actually a transparent filament, which is preferable to clear filament showing as white.

spuder, 2025-01-09:
Great observation. 

This is the line of code where transparency is hard coded to `ff`

https://github.com/spuder/OpenSpool/blob/main/firmware/bambu.h#L131

This will need to be changed to a variable for alpha. 

I took a quick look at Bambu Studio and I see the way they are handling it is by making a new filament with name 'translucent'. 
It appears they currently only support PETG, there is no translucent PLA/ABS for example. Transparent PETG has its own code GFG01

https://github.com/bambulab/BambuStudio/blob/c1b3cc0bf1c6c9be26ffdaa772da5a2716751677/resources/profiles/BBL/filament/Bambu%20PETG%20Translucent%20%40base.json#L3
[image]

mytechguyri, 2025-01-14:
The thing is... I believe BambuStudio accepts the Alpha channel variable... You're passing FF as default according to the logs in the mqtt data... 
17:09:16
[I]
[mqtt:148]
Publishing {"print":{"sequence_id":"0","command":"ams_filament_setting","ams_id":255,"tray_id":254,"tray_color":"E27A2BFF","nozzle_temp_min"

which is fine.. but if the end user passes an alpha channel value in the NFC tag, Openspool chokes on it and logs the following error "17:11:28
[E]
[bambu:122]
Invalid color_hex length (expected 6 characters)"

Better  would be... if len = 6  (i.e. E27A2B) then append FF and pass E27A2BFF in the mqtt data
if len = 8 then pass the 8 digits as is in the mqtt data, which should pass the alpha channel value.

There are cases in Bambu Studio where it allows you to select transparent filament, without choosing the "Translucent" type... have you ever seen where the folor shows as a grey and white checkboard?   That's because BambuStudio somehow has a 00 alpha channel value for the color... They don't make it easy to input, but the software does handle it just fine if the value is set to 00

raihei, 2025-01-17:
> There are cases in Bambu Studio where it allows you to select transparent filament, without choosing the "Translucent" type... have you ever seen where the folor shows as a grey and white [...]

Somehow I set it to transparent in the past for my transparent filament, but now I don't know anymore, how I did that. Can you give me a hint, how to set it in BambuStudio?

raihei, 2025-01-20:
@spuder I prepared a commit to include creation of tags with filament color alpha channel, if wanted I can create a pull request

[image]

spuder, 2025-01-21:
Yes please