spuder/OpenSpool #24: Proper RFID Programming

usajet, 2025-01-13:
What are the proper steps to program the RFID Tags for Bambu Labs X1C? Just having issues with the programming.

spuder, 2025-01-13:
Sorry to hear you're having trouble programming tags. 

Is it unreliable or what is the problem you were finding? 

Program a tag you go into the web interface at http://openspool-xxxxxx.local or http://192.168.x.x

Then you need to select the filament type filament color and filament brand and filament temperature range. 

After you said that press the right NFC button and then touch tag to the reader. It works best in the corners of the reader, hold it for one to two seconds.

If there's a problem writing, it will light up red and you'll see an error in the logs

usajet, 2025-01-13:
Okay that is what I needed. Wasn't sure. I try to scan the rfid on my phone and no info shows up. I have a android phone.

usajet, 2025-01-13:
Can we add our own filament to the program?

spuder, 2025-01-13:
Currently, it only supports a select, subset of bambu recognized codes. If it's not a filament that is currently recognized just select generic.

There are plans to improve this and pull from other databases, but for now it's only the bambu codes

mytechguyri, 2025-01-14:
If we put a brand (because I use the NFC tags for more than just scanning to the printer, I use them to inventory my filament as well)   So, if I put brand as "Jayo" or "Flashforge" or "Reprapper"... will the device just treat that as generic or will it error because its not one of the defined brands?

spuder, 2025-01-15:
get_bambu_code()

It will return an empty string which the printer will probably ignore (unverified). 

Would you be willing to open a feature request to change that behavior? We could make it fall back to `generic` instead of failing silently.