spuder/OpenSpool #27: Internal LED pinout is backwards

spuder, 2025-01-14:
The internal led lights up the wrong colors (red and green are swapped)

Also it should probably be hidden from the UI as it is not used

raihei, 2025-01-15:
I was already wondering, for what the internal LED was used, because I couldn't find any references in the sources.
On my board (Waveshare ESP32-S3-Zero) the colors are correct. On which board did you discover that issue?

impeeza, 2025-11-08:
> I was already wondering, for what the internal LED was used, because I couldn't find any references in the sources. On my board (Waveshare ESP32-S3-Zero) the colors are correct. On which board did you discover that issue?

Hello there,  I only could get a Waveshare ESP32-S3-Zero, and I am very confused if I can use it for this project. the GPIO pins 34, 36 AND 44 Are not exposed on the zero and the GPIO pins 14, 16, & 38 are small test points.

do I need to create a YML for the zero and build from code?

are you using the zero for the project?  

thanks for your help.