spuder/OpenSpool #30: Make mqtt client id unique

spuder, 2025-01-14:


raihei, 2025-01-29:
I tried it, but looks like it is not possible. Or is there another way?

[image]

spuder, 2025-01-29:
Thanks for trying, if it isn't templateable then it likely isn't possible. 
I ran into the same issue with the mqtt broker address not being templatable. 

https://discord.com/channels/429907082951524364/524177279270780928/1300834004718587998

That was fixed in https://github.com/esphome/esphome/pull/7716

I suspect it might take a pull request to esphome community to fix this. 

Feel free to take this on if you are interested. As it isn't really going to break anything I"m going to mark as wont-fix for now.