spuder/OpenSpool #32: Everything seems to be working (I think), but printer is not updated.

mytechguyri, 2025-01-15:
Running 1.16 firmware.... it scans the NFC tag, publishes the change.... nothing changes on the printer
A1 Mini

(you need a support email address :) )
`19:20:06
[D]
[pn532:367]
Mifare ultralight
19:20:06
[D]
[NFC:064]
Payloa
19:20:06
[D]
[NFC:131]
Input strin
19:20:06
[I]
[mqtt:150]
Publishing {"print":{"sequence_id":"0","command":"ams_filament_setting","ams_id":255,"tray_id":254,"tray_color":"FFFFFFFF","nozzle_temp_min":205,"nozzle_temp_max":215,"tray_type":"PLA","setting_id":"","tray_info_idx":"GFL99"}}
19:20:06
[D]
[pn532:200]
Found new tag '04-AD-D4-6C-BB-2A-81'
19:20:06
[D]
[pn532:204]
NDEF formatted records:
19:20:06
[D]
[pn532:206]
application/json
19:20:06
[D]
[pn532:379]
Waiting to read next tag
19:20:06
[W]
[component:237]
Component pn532 took a long time for an operation (198 ms).
19:20:06
[W]
[component:238]
Components should block for at most 30 ms.
19:20:08
[D]
[NFC:128]
Input string is empty`

mytechguyri, 2025-01-15:
A1 Mini with firmware 1.04.00.00

raihei, 2025-01-15:
You can connect to the printer via MQTT Explorer and check what it reports:

https://mqtt-explorer.com/

Please use the settings shown here to connect:

https://forums.homeseer.com/forum/lighting-primary-technology-plug-ins/lighting-primary-technology-discussion/mcsmqtt-michael-mcsharry/1601281-how-to-integrate-bambu-lab-3d-printer-via-mqtt

Password to connect is the access code, can be found in the printer menu.

Wait until a very long report is shown, then you can stop the updating (pause button in the upper bar in MQTT explorer) and on the right side, you can see the contents of the report.

[image]

spuder, 2025-01-15:
Likely culprit: https://github.com/spuder/OpenSpool/issues/28

The "upload settings" button is intended for debugging and it looks like it is sending the temperature values incorrectly. 
The nfc tag works as expected.

mytechguyri, 2025-01-16:
Okay, wasn't sure if it was a problem with my NFC tag or what... so tag is working as expected.... other than its not updating the printer as expected, so there's that.. lol.   just out of curiousity... Are you attempting the updates via local MQTT on the printer, or CLOUD MQTT with Bambu?    If I'm not mistaken, Bambu has essentially disabled local MQTT on cloud connected printers, and requires you send through the Bambu Cloud MQTT server...  Perhaps that's why its not updating?

mytechguyri, 2025-01-16:
Well.... I don't know what I did, but today its working as it should... scan the tag, color changed on the printer.   All i can figure is perhaps (and this is a complete wild ass guess)... Between Bambu Handy, an instance of Bambu Studio, an instance of Orca Slicer,  and the Panda Touch all being connected and operating, we reached the connection limit for the printer?   I dunno, but its working.

closed 2025-01-16 by mytechguyri

spuder, 2025-01-16:
Great! Glad to hear. 

We don't know where the limit is yet. It might be 4 MQTT connections, but haven't confirmed that yet. It's entirely possible it was getting rejected.