spuder/OpenSpool #34: Android Support of NFC Tag shortcut

spuder, 2025-01-15:
The openspool kits come with a tag pre-programmed to open the bambu handy app on iPhone. It does that by using a custom URL `bambu://`

Figure out what the equivalent url is for android phones

- [ ] Document what the url is
- [ ] Research if possible to have esp32 program tags through the `extras` section of the web interface. 

The url is likely something like this: 

`android://bbl.intl.bambulab.com` or `vnd.android.nfc://ext/android.com:pkg`

spuder, 2025-01-16:
Android needs an AAR with value `bbl.intl.bambulab.com`

Additional information here: https://discord.com/channels/1298381115706576907/1298381115706576910/1329449638578618460

Thanks to Rich in Discord for finding this.

mytechguyri, 2025-01-17:
If its any help, I wrote a tag to open Bambu Handy using NFC Tools on my phone, here's what the tag contains.

[image]