spuder/OpenSpool #36: Firmware Update Introducing New Authorization Control System

koffienl, 2025-01-16:
https://blog.bambulab.com/firmware-update-introducing-new-authorization-control-system-2/

As part of our ongoing commitment to enhance the overall security of our products, we are introducing an authorization and authentication protection mechanism for the connection and control of Bambu Lab 3D printers. This step is a significant security enhancement to ensure only authorized access and operations are permitted.

Critical Operations That Require Authorization
The following printer operations will require authorization controls:

Binding and unbinding the printer.
Initiating remote video access.
Performing firmware upgrades.
Initiating a print job (via LAN or cloud mode).
Controlling motion system, temperature, fans, **AMS settings**, calibrations, etc.

mytechguyri, 2025-01-16:
This is NOT good... Don't update to this new firmware.... I expect its to break Panda Touch, as well as OpenSpool.... and they add another layer to OrcaSlicer, so you can no longer seemlessly send your prints, you have to save the gcode, then use Bambu Connect software to send the gcode to the printer.... Make your dissatisfaction with this known to Bambu (I'm also thinking of lodging a complaint with the FTC about this anti-competitive tactic that Bambu is introducing under the false premise of "security")

spuder, 2025-01-16:
I've already been in contact with Bambu's development group, I'm waiting to hear back. 

While this is an unfortunate setback, it's not the end of the road for OpenSpool. There are still multiple paths forward. Some involve software changes, others involve hardware changes. 

When Bambu made changes to disable the LAN mode, they took the feedback to heart and backpedaled so that LAN mode is still an option. I'm optimistic they will do the same here. 

The list of current options

1. Document users should not update firmware (P1/A1 are not enforcing yet)
2. Leverage Bambu Connect as mqtt proxy. This might be a good option anyway since there is a device limit. By leveraging the Bambu Connect as a REST to MQTT proxy, 3rd party devices might still be supported 
3. Work with bambu to get TLS signing certificate pipeline for firmware. This might prevent end users from compiling their own unsigned esphome configs, but there likely will be a path forward here. 
4. Resdesign the hardware to plug directly into the AMS interface CANBUS. While tricky, it is doable and would be really hard for them to block. 

In the mean time I'm leveraging all the channels I have to talk with Bambu about how this will hurt 3rd party products like OpenSpool, X-Touch, Panda Touch, BLLED, HomeAssistant

mytechguyri, 2025-01-16:
Well, to help them "backpedal" I filed my complaint with the FTC about their anti-competitive tactic disquised as "security" (its already secure through use of the access token system, which is only available on the printer itself... they know this is not about security, its about shutting down third party devices.

FTC Report Number
182500022

xperiments, 2025-01-17:
@spuder Please contact me at xperiments@gmail.com

darkorb, 2025-01-18:
> this anti-competitive tactic that Bambu is introducing under the false premise of "security"

100% this is a false thing from them, there is zero need to do this on LAN. I couldn't care less if they tightend up the cloud side of things (which they probably should do, if what they say about the number of unappreoved requests is true) but targeting local network use is just using insane nonsense arguments like "well your network is also insecure because trojans" sigh. Like, if my network had a trojan and someone wanted to target the printer, they'd just abuse the machine that connect is on (not to mention, they'd rather that they just snaffle your banking/online account creds instead tbh).

You can also open support cases under `Complaints and Suggestions -> Complaints -> Product issues` and see if they listen. Doubt it though.

Schrolli91, 2025-01-20:
There are some clarifications from Bambu Lab now
https://blog.bambulab.com/updates-and-third-party-integration-with-bambu-connect/

spuder, 2025-01-20:
Thanks for the link.

The new developer mode will likely be the best path forward.
We are also looking at a design that uses a different interface. Holding
off on details until a prototype exists.

luckyzor, 2025-01-22:
Sorry by asking this here, but I cant find any information.
What is the last firmware supported by openspool for a X1C?
Is it the 1.0.8?

I want to know so I can downgrade and lock internet access do the printer.
Thank you

spuder, 2025-01-22:
The latest firmware that came out last week 1.08 breaks the MQTTAPI for X one users. 

X one users will either need to downgrade to 107 or wait for the new developer mode that will be coming later this year From bambu labs

akagalford, 2025-01-22:
I'm currently running Openspool 1.17 with Firmware 1.08.02.00 on my X1c and
the NFC scan for updating my 5th spool settings change on my X1c is working
for me. What part is having issues?

luckyzor, 2025-01-22:
> The latest firmware that came out last week 1.08 breaks the MQTTAPI for X1 users.
> 
> X1 users will either need to downgrade to 1.07 or wait for the new developer mode that will be coming later this year from bambu labs

I'm using orca slicer and Home assistant with 1.08.02 and thats why I asked this question, because I saw in the "buy kit" page the last firmware supported was 1.07...

luckyzor, 2025-01-23:
> The latest firmware that came out last week 1.08 breaks the MQTTAPI for X1 users.
> 
> X1 users will either need to downgrade to 1.07 or wait for the new developer mode that will be coming later this year from bambu labs

Can you confirm this? I really want to use this project.

koffienl, 2025-01-23:
> > The latest firmware that came out last week 1.08 breaks the MQTTAPI for X1 users.
> > X1 users will either need to downgrade to 1.07 or wait for the new developer mode that will be coming later this year from bambu labs
> 
> Can you confirm this? I really want to use this project.

As far as I know is the firmware still in Beta.
The beta firmware with the security 'feature' is 01.08.03.00
The latest stable/public firmware version is 01.08.02.00 and does not have this issue.