spuder/OpenSpool #39: Anycubic Support?

SquidBytes, 2025-01-20:
I was researching around for a system similar to this and discovered this project.

I have the Anycubic Kobra 3, which uses an ACE Pro with RFID (AMS clone)

I'm curious on what would be needed - or how difficult it would be to implement support for Anycubic.

raihei, 2025-01-21:
From the first look, it seems, Anycubic only allows MQTT access to the printer via their cloud.
Or is there a mode where a direct MQTT connection via IP and username/password authentication to the printer is possible?

simon314159, 2025-03-23:
I'm also interested in an Anycubic integration. I found a project which adds a Home Assistent integration via MQTT and the LAN mode, only IP ist needed: https://github.com/metheos/anycubic-s1-mqtt-bridge