spuder/OpenSpool #52: Support for TigerTag

spuder, 2025-03-01:
Add support for Tiger Tag

https://tigertag.io

[image]

TigerTag-Project, 2025-07-16:
Hi @spuder ,

Thanks for your interest in integrating TigerTag!

To get started, you can refer to the official TigerTag RFID Guide available here:
📘 https://github.com/TigerTag-Project/TigerTag-RFID-Guide

The most important step is to include the TigerTag ID database, which allows decoding the tag data properly:
📁 https://github.com/TigerTag-Project/TigerTag-RFID-Guide/tree/main/database
This database can be used locally or accessed via a free and anonymous REST API.

We’re fully available to support this integration — feel free to reach out anytime, we’re happy to help!

— TigerTag Team

goeland86, 2025-08-05:
@spuder I'm happy to try and take a crack at implementing this. I have TigerTag NFCs and a USB NFC reader. I'll get my hands on a PN532 as well shortly.