spuder/OpenSpool #55: Connection to A1 MQTT not possible

Ciredm, 2025-03-22:
Hi!
I've big issue to use OpenSpool, I'm unable to connecte to MQTT:

```

[I][mqtt:250]: Connecting to MQTT...
[V][mqtt.idf:121]: Event dispatched from event loop event_id=7
[V][mqtt.idf:124]: MQTT_EVENT_BEFORE_CONNECT
[V][mqtt.idf:121]: Event dispatched from event loop event_id=0
[E][mqtt.idf:162]: MQTT_EVENT_ERROR
[E][mqtt.idf:164]: Last error code reported from esp-tls: 0x8017
[E][mqtt.idf:165]: Last tls stack error number: 0x7f00
[E][mqtt.idf:166]: Last captured errno : 0 (Success)
[V][mqtt.idf:121]: Event dispatched from event loop event_id=2
[V][mqtt.idf:133]: MQTT_EVENT_DISCONNECTED
[I][main:783]: Disconnected from printer!
[W][mqtt:340]: MQTT Disconnected: TCP disconnected.
[I][mqtt:250]: Connecting to MQTT...
[V][mqtt.idf:121]: Event dispatched from event loop event_id=7
[V][mqtt.idf:124]: MQTT_EVENT_BEFORE_CONNECT
[V][mqtt.idf:121]: Event dispatched from event loop event_id=0
[E][mqtt.idf:162]: MQTT_EVENT_ERROR
[E][mqtt.idf:164]: Last error code reported from esp-tls: 0x8017
[E][mqtt.idf:165]: Last tls stack error number: 0x7f00
[E][mqtt.idf:166]: Last captured errno : 0 (Success)
[V][mqtt.idf:121]: Event dispatched from event loop event_id=2
[V][mqtt.idf:133]: MQTT_EVENT_DISCONNECTED
[I][main:783]: Disconnected from printer!
[W][mqtt:340]: MQTT Disconnected: TCP disconnected.

```

I can connect to A1 MQTT directly using MQTT Explorer, access code are correct.
This error seems to be known (https://esp32.com/viewtopic.php?t=28325). Bambu Lab public key is not include in OpenSpool code?

Thanks for your help.

spuder, 2025-03-22:
Sorry to hear it isn't working. OpenSpool has been tested on multiple A1 Mini and P1S. 

What MCU (esp32-s2/ esp32-s3) ? 
What firmware version of A1 ? 

Others have reported occasional issues connecting to printers, but it usally is resolved by disconnecting all MQTT clients and then rebooting the printer. 

Lastly double check the serial number and LAN Access code are correct (Including case sensitive).

Ciredm, 2025-03-22:
Hi, thank for your very quick answer!
 I use an ESP32-S2. My A1printer have 01.04.00.00 firmware version.

For access code and LAN access, I've retry lot of time since 3 days (and it is working using MQTT Client to connect to printer).
I've reboot twice my printer, I'l try with my panda knomi disconnected.

spuder, 2025-03-27:
What firmware version are you using? 

The esp32-s2 barely has enough memory to establish TLS connections. 
Also there are fake esp32-s2 boards which do not have PSRAM which make this problem more pronounced. 
Firmware 1.19 removed some features to free up a few bytes on the s2 boards. 

While you probably don't want to upgrade, the s3 boards have sufficient memory and are more reliable at connecting. All new openspool boards from the store have switched to the s3 for that reason. 

Do you see in the web interface how much heap is available? 

I'll borrow an A1 printer to test and see if I can reproduce.

Ciredm, 2025-03-29:
Hi,
I've clean all my config and re-install esphome and openspool yesterday (28/03/25)
ESPHome 2025.3.2 and main branch for openspool git (identified in code  by v1.19.1)

Only change in source code is: led-external.yaml @line 13: commented due to compiling error ESP-IDF 5.
`
$esphome run lolin_s2_mini.yaml 
INFO ESPHome 2025.3.2
INFO Reading configuration lolin_s2_mini.yaml...
WARNING The selected ESP-IDF framework version is not the recommended one. If there are connectivity or build issues please remove the manual version.
WARNING GPIO0 is a strapping PIN and should only be used for I/O with care.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
WARNING The selected ESP-IDF framework version is not the recommended one. If there are connectivity or build issues please remove the manual version.
INFO Generating C++ source...
INFO Updating https://github.com/espressif/esp-protocols.git@mdns-v1.8.0
INFO Compiling app...
Processing openspool (board: lolin_s2_mini; framework: espidf; platform: platformio/espressif32@6.9.0)
---------------------------------------------------------------------------------------------------------
HARDWARE: ESP32S2 240MHz, 320KB RAM, 4MB Flash
 - framework-espidf @ 3.50301.0 (5.3.1) 
 - tool-cmake @ 3.16.4 
 - tool-ninja @ 1.7.1 
 - tool-riscv32-esp-elf-gdb @ 12.1.0+20221002 
 - tool-xtensa-esp-elf-gdb @ 12.1.0+20221002 
 - toolchain-esp32ulp @ 2.35.0-20220830 
 - toolchain-riscv32-esp @ 13.2.0+20240530 
 - toolchain-xtensa-esp-elf @ 13.2.0+20240530
Reading CMake configuration...
Dependency Graph
|-- ArduinoJson @ 6.18.5
Compiling .pioenvs/openspool/src/main.cpp.o
Linking .pioenvs/openspool/firmware.elf
RAM:   [=         ]  11.2% (used 36844 bytes from 327680 bytes)
Flash: [=======   ]  69.2% (used 1270339 bytes from 1835008 bytes)
Building .pioenvs/openspool/firmware.bin
Creating esp32s2 image...
Successfully created esp32s2 image.
esp32_create_combined_bin([".pioenvs/openspool/firmware.bin"], [".pioenvs/openspool/firmware.elf"])
SHA digest in image updated
Wrote 0x1463b0 bytes to file /home/mederic/OpenSpool/firmware/.esphome/build/openspool/.pioenvs/openspool/firmware.factory.bin, ready to flash to offset 0x0
esp32_copy_ota_bin([".pioenvs/openspool/firmware.bin"], [".pioenvs/openspool/firmware.elf"])
===================================== [SUCCESS] Took 31.21 seconds =====================================
INFO Successfully compiled program.
Found multiple options for uploading, please choose one:
  [1] /dev/ttyACM0 (ESP32-S2)
  [2] Over The Air (openspool.local)
(number): 1
esptool.py v4.8.1
Serial port /dev/ttyACM0
Connecting...
Chip is ESP32-S2FNR2 (revision v1.0)
Features: WiFi, Embedded Flash 4MB, Embedded PSRAM 2MB, ADC and temperature sensor calibration in BLK2 of efuse V2
Crystal is 40MHz
MAC: cc:8d:a2:82:12:f0
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
Configuring flash size...
Auto-detected Flash size: 4MB
Flash will be erased from 0x00010000 to 0x00146fff...
Flash will be erased from 0x00001000 to 0x00006fff...
Flash will be erased from 0x00008000 to 0x00008fff...
Flash will be erased from 0x00009000 to 0x0000afff...
Compressed 1270704 bytes to 846916...
Wrote 1270704 bytes (846916 compressed) at 0x00010000 in 9.4 seconds (effective 1086.7 kbit/s)...
Hash of data verified.
SHA digest in image updated
Compressed 22400 bytes to 13786...
Wrote 22400 bytes (13786 compressed) at 0x00001000 in 0.3 seconds (effective 665.3 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 134...
Wrote 3072 bytes (134 compressed) at 0x00008000 in 0.0 seconds (effective 552.5 kbit/s)...
Hash of data verified.
Compressed 8192 bytes to 31...
Wrote 8192 bytes (31 compressed) at 0x00009000 in 0.1 seconds (effective 801.6 kbit/s)...
Hash of data verified.

Leaving...
INFO Upload with baud rate 460800 failed. Trying again with baud rate 115200.
esptool.py v4.8.1
Serial port /dev/ttyACM0
Connecting....
Chip is ESP32-S2FNR2 (revision v1.0)
Features: WiFi, Embedded Flash 4MB, Embedded PSRAM 2MB, ADC and temperature sensor calibration in BLK2 of efuse V2
Crystal is 40MHz
MAC: cc:8d:a2:82:12:f0
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Auto-detected Flash size: 4MB
Flash will be erased from 0x00010000 to 0x00146fff...
Flash will be erased from 0x00001000 to 0x00006fff...
Flash will be erased from 0x00008000 to 0x00008fff...
Flash will be erased from 0x00009000 to 0x0000afff...
Compressed 1270704 bytes to 846916...
Wrote 1270704 bytes (846916 compressed) at 0x00010000 in 9.4 seconds (effective 1085.2 kbit/s)...
Hash of data verified.
SHA digest in image updated
Compressed 22400 bytes to 13786...
Wrote 22400 bytes (13786 compressed) at 0x00001000 in 0.3 seconds (effective 671.3 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 134...
Wrote 3072 bytes (134 compressed) at 0x00008000 in 0.0 seconds (effective 616.9 kbit/s)...
Hash of data verified.
Compressed 8192 bytes to 31...
Wrote 8192 bytes (31 compressed) at 0x00009000 in 0.1 seconds (effective 791.9 kbit/s)...
Hash of data verified.

Leaving...
Error: ESP32-S2FNR2 (revision v1.0) chip was placed into download mode using GPIO0.
esptool.py can not exit the download mode over USB. To run the app, reset the chip manually.
To suppress this note, set --after option to 'no_reset'.
`

Error is still here!

Ciredm, 2025-05-10:
Hi,

Using S3 mini, no more issue. I suppose (but not tested), issue is due to Bambu A1 mqtt connexion limit (near 4 clients) with long long time release (hours, maybe 1 day). 

1) Home Assistant
2) Panda Knomi
3&4) Bambu Studio (2 computers, depend of my location)

I cannot connect Openspool... But when received S3 board an try it without open bambu studio, OpenSpool's mqtt  work.
And one bambu studio return -1 error when try to contact A1.

I havn't performed more tests.

Bye.

closed 2025-07-03 by spuder