spuder/OpenSpool #56: Simple Docs

mworsnop, 2025-03-26:
I've got my openspool preassembled and tested, so I didn't have to mess up making it work. I got it on the network. Everything's great. 
But I'm old and I guess I just don't understand every single step here. It manages to read an RF ID and I can see that. It connected to my bambu X1C printer. Now I'm trying to figure out exactly what I can use it for and how to use it. I would've loved to sent a message to Support or something, but I have no idea how to contact anybody. Maybe somebody can give me an email address or a support form or something. Sorry for the stupid questions.

spuder, 2025-03-27:
I'm glad to hear you got an openspool and have it connected to your Bambu X1C. 

As far as usage there are multiple ways to use it. 

The intended use case is to put the stickers on the spools then either use the openspool mini or the iPhone/android app to read the tags. 

Were you are likely getting hung up is by the AMS. The current firmware only works with the external spool, but there is code in works to make it also work with an AMS aswell.

mworsnop, 2025-03-27:
I assume that I use the open spool app to write the information to the
stickers. Is that correct?

And the information that gets written is what is on the first part of the
app.

mworsnop, 2025-03-30:
Would it be possible to add a field up at the top that would be like an ID or something that I could enter manually before I create a tag. Then later, when I grab the spool, I can scan it and make sure that number is the number that's matching in my database. Does that make sense? Is that possible?

Thing maybe you could clear up I don't understand what they do. In the filament settings at the top, there is a button to upload settings. Where is it uploading it too and what is the purpose of it?

Then down at the bottom in the extra section, there is a bamboo handy create tag button. What does it do? And I assume the open school create tag will write to a blank tag or something. I'm not sure. I'm sorry I'm really blind here trying to figure this out. Sure would appreciate some help. Thank you.

raihei, 2025-04-11:
> Would it be possible to add a field up at the top that would be like an ID or something that I could enter manually before I create a tag. Then later, when I grab the spool, I can scan it and make sure that number is the number that's matching in my database. Does that make sense? Is that possible?
> 
> Thing maybe you could clear up I don't understand what they do. In the filament settings at the top, there is a button to upload settings. Where is it uploading it too and what is the purpose of it? And directly under that it says right NFC. What's the difference between that and the open spool create tag down at the bottom?
> 
The "Upload settings" button uploads the currently selected filament settings to the printer. That button is for testing only and might not work 100% (There were/are problems with the temperatures)
The "Write NFC" button will write a tag with the currently selected filament settings.

> Then down at the bottom in the extra section, there is a bamboo handy create tag button. What does it do? And I assume the open school create tag will write to a blank tag or something. I'm not [...]

The "BambuHandy Create Tag" will write a tag, that, when scanned with a mobile device, opens BambuHandy on that device (if installed)

The "OpenSpool Create Tag" will write a tag, that, when scanned with a mobile device, opens the browser with the website of your openspool device.